cartographer-project/cartographer_ros #1710: Can't see the tracked pose as a topic

ArghyaChatterjee, 2022-03-12:
Hi, 
I am trying to get tracked pose from cartographer. I am using ros melodic and ubuntu 18.04 where cartographer and cartographer ros binary packages are installed through apt. I am using a single 2d rplidar A1M8.

As suggested in this [pr](https://github.com/cartographer-project/cartographer_ros/pull/1099) and [here](https://github.com/cartographer-project/cartographer_ros/blob/master/docs/source/ros_api.rst), if I set the parameter `publish_tracked_pose = true`, I get an error saying `lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'publish_tracked_pose' was used the wrong number of times.` Here is my .lua file:
```
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 10,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
 
  publish_tracked_pose = true,

  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 7
TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 10
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1

return options
```
As suggested by some issues, I changed the parameter `publish_tracked_pose = on`, the error was gone but I can't see the `tracked_pose` topic published. 

Any help is appreciated.

eithwa, 2022-03-21:
I found that version 1.0.0 does not contain this feature, you may need to update the version.

goldtimes, 2022-11-23:
> 
i have question, 1.0.0 version, is latest. how can i update this version?

eithwa, 2025-04-07:
```
publish_tracked_pose = true,
```
Add it to the option field of the lua you are using and restart the program