cartographer-project/cartographer_ros #1713: Tuning on my own pure-laser dataset (TSDF tuning help most needed)

Enigmatisms, 2022-03-29:
Recently I have been working on my own dataset generated by simulation. For most of the bags (out of 20 bags with 5 scenes), probability-grid-based cartographer yields good results in the offline node, yet for some of the scenes, when the angular resolution of the simulation is set to be 0.1 deg, cartographer failed to generate valid maps, see the following figures:
[image]
[image]
I have tried various parameter configurations (all in correlative scan matching since ceres matcher won't work at all), including modifying the linear and angular search range, weights, and the probability of hitting and missing, yet they all failed. Links to my rosbags are below, they both have 240 deg fov and 0.1 deg angular resolution, and both of them have scan data only.
- [Google drive:first_bag](https://drive.google.com/file/d/1rLeXSBqOX_SCbxcJ2XS-9II5zft9ks_d/view?usp=sharing)
- [Google drive:second_bag](https://drive.google.com/file/d/1-LvFnon0pvKFqPxJJ9FuF5XR5pfZC0oK/view?usp=sharing)
- If you are suspicious about the way these bags are generated, you can refer to [Github: LiDARSim2D](https://github.com/Enigmatisms/LiDARSim2D)
Also, when I tried to use TSDF as map representation, none of the 20 bags worked with cartographer. Though I tried to tune some of the params given, the absence of documentation makes me try without direction. The picture listed below is currently the best result tuned, yet it is far from satisfactory:
[image]

- Rosbag can be found in [Google drive:tsdf_test.bag](https://drive.google.com/file/d/1XcaZSnX7yX95axu46Eo8cDCNpEy5KIbv/view?usp=sharing)
- Configuration file for probability-grid-based SLAM: [Google drive:my_robot.lua](https://drive.google.com/file/d/1ad-E47jCeNTP3Z8bWKkbdGB4frromgto/view?usp=sharing), [Google drive:trajectory_builder_2d.lua](https://drive.google.com/file/d/1wgl54JE1mPkTuHZGvH9DDBrsVY3bXZCl/view?usp=sharing). For TSDF, there is no need to provide configuration files since... none of trails worked. 

Would appreciate useful help of any kind.

eithwa, 2022-04-12:
[image]
try this one
https://gist.github.com/eithwa/1815bab76f6007c826d7e2e28b2d4159

Enigmatisms, 2022-04-12:
> [image] try this one https://gist.github.com/eithwa/1815bab76f6007c826d7e2e28b2d4159

Thanks! This result is awesome and reproducible. I always thought that rotation is harder to handle therefore omitting the translational weight.

closed 2022-04-12 by Enigmatisms