cartographer-project/cartographer_ros #1715: using livox mid 70+IMU+wheel odometry can not work

gongyue666, 2022-04-14:
Hi, I want to using cartographer_ros with livox mid 70+IMU+wheel odometry for 3D SLAM, but it even can not run normally. could anyone work with that, and can share your launch file and .lua file? Thanks

eithwa, 2022-05-03:
Can you provide the bag, lua, urdf, launch files you used?

gongyue666, 2022-05-09:
Hi, thanks for reply, I will upload them today

gongyue666, 2022-05-09:
[https://drive.google.com/file/d/1xLmrmYJ5-YJHvr2ZjGvs1Y6Vaa2Ysc16/view?usp=sharing](url)

gongyue666, 2022-05-09:
please check the download link, thanks

eithwa, 2022-05-09:
[image]

[image]

topic "/robot_pose_ekf/odom_efk" has no child_frame_id, so I use topic "odom"

The num_accumulated_range_data setting is too large, I changed it and added the second lidar
```
TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 5
```

https://gist.github.com/eithwa/a803f943a53faa3c5e1deab948161f48

gongyue666, 2022-05-10:
Hi thanks so much for your reply, but I still can not get your result in my robot, even using your files. The log is like below:  "[ WARN] [1652145293.986301439, 1650071628.187052015]: W0510 09:14:53.000000  7251 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1652145294.126033430, 1650071628.328561289]: W0510 09:14:54.000000  7251 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1652145294.251899361, 1650071628.449899706]: W0510 09:14:54.000000  7251 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1652145294.392039288, 1650071628.591239616]: W0510 09:14:54.000000  7251 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1652145294.519905193, 1650071628.722573452]: W0510 09:14:54.000000  7251 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1652145294.650041398, 1650071628.843433022]: W0510 09:14:54.000000  7251 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1652145294.652907557, 1650071628.853543574]: Could not compute submap fading: "map" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1652145294.785903181, 1650071628.984647767]: W0510 09:14:54.000000  7251 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)

gongyue666, 2022-05-10:
and the tracking_frame is set by "base_link", should is be "imu_link"?

eithwa, 2022-05-10:
> Hi thanks so much for your reply, but I still can not get your result in my robot, even using your files. The log is like below: "[ WARN] [1652145293.986301439, 1650071628.187052015]: W0510 [...]

set this in robot.lua
```
num_point_clouds = 2,
```
I haven't tried imu_link, I usually use base_link

gongyue666, 2022-05-10:
hi I use this file like this, and it worked:
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "imu_link",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 2,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1,
  landmarks_sampling_ratio = 1.,
}

TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 5

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 320
POSE_GRAPH.constraint_builder.sampling_ratio = 0.03
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

return options

gongyue666, 2022-05-10:
thanks for your reply, again~

gongyue666, 2022-05-11:
sorry, I face some new problems:1 The 2D grid map seems "dirty", how can I chang it to a better map like using a 2D laser scan build？2 when I using cartographer in "pure localization" state, I think it is still update the prior map ( both piont cloud and 2d Grid Map). is it a normal state? how can I stop the update? here is my config files:

gongyue666, 2022-05-11:
https://drive.google.com/file/d/145gZcj4zLUOlB-Sf12J09SDdt307hflV/view?usp=sharing

eithwa, 2022-05-27:
I think it's because there aren't enough points to hit the ground, causing the map you said to look "dirty", which can be improved with 2d mapping.
When pure localization is used, several submaps will be retained, which makes the map still change. When the robot moves, the old submap will be discarded and restored to the original map. The number of submaps can be modified in lua.
https://github.com/cartographer-project/cartographer_ros/blob/c06879b63567a78ef92b7d1fa79453839e36ffec/cartographer_ros/configuration_files/backpack_2d_localization.lua#L18