cartographer-project/cartographer_ros #1721: cartographer with rotating VLP-16

nonlinear1, 2022-05-23:
I have made a rotating velodyne lidar device with vlp16, gimbal motor , slip ring, imu, absolute encoder. I want to run cartographer with my rotating velodyne lidar device. I have not some experineces on rotating velodyne to run cartographer. So could someone give me some suggestion? Thank you very much. the following is my rotating velodyne device

https://user-images.githubusercontent.com/19766939/169825881-8b48dfae-4a8b-45b9-b0ac-f1f7b9c015cc.mp4

:

eithwa, 2022-05-27:
you can use the encoder to transform the base_link to the tf data of the lidar frame, you can refer to odometry
http://wiki.ros.org/navigation/Tutorials/RobotSetup/Odom

nonlinear1, 2022-05-27:
I cann't understand. the base_link can stand still and the lidar frame is rotating.

eithwa, 2022-06-01:
creat tf tree link
base_link->motor_link->laser_link
and use motor encoder data to transform motor_link and that laser_link will rotate

nonlinear1, 2022-06-01:
Thank you, I will try.