cartographer-project/cartographer_ros #1726: Error while installing dependencies: libabsl-dev not available

fjs-acc, 2022-07-03:
I tried installing cartographer ros on a newly set up ubuntu system (20.04.4  LTS with ros noetic installed) and followed the following steps of the documentation:
`sudo apt-get update`
`sudo apt-get install -y python3-wstool python3-rosdep ninja-build stow`

`mkdir catkin_ws`
`cd catkin_ws`
`wstool init src`
`wstool merge -t src https://raw.githubusercontent.com/cartographer-project/cartographer_ros/master/cartographer_ros.rosinstall`
`wstool update -t src`

`sudo rosdep init`
`rosdep update`

Up until this point everything worked fine, but when I try the next step

`rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y`

I get the following error message:

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
cartographer: [libabsl-dev] defined as "not available" for OS version [focal]
`
Does anyone know what is causing this issue and how I can fix it?

christopherbohn, 2022-07-04:
Same issue here - ROS noetic on Ubuntu 20.04.4 LTS

fjs-acc, 2022-07-04:
Well, as it turns out that this dependency was added to the cartographer package roughly a week ago (as can be seen here [https://github.com/cartographer-project/cartographer/pull/1875](url)); removing the respective line from the package.xml  and installing abseil via the command from the guide (ie. `src/cartographer/scripts/install_abseil.sh`) seems to have fixed the issue, atleast in so far as it allowed me to build and install cartographer ros. 

I'm not 100% sure if it

KimHyung, 2022-07-05:
> Well, as it turns out that this dependency was added to the cartographer package roughly a week ago (as can be seen here [https://github.com/cartographer-project/cartographer/pull/1875](url)); removing the respective line from the package.xml and installing abseil via the command from the guide (ie. `src/cartographer/scripts/install_abseil.sh`) seems to have fixed the issue, atleast in so far as it allowed me to build and install cartographer ros.
> 
> I'm not 100% sure if it fixed the issue without creating new ones though.
> 
> PS: I accidentally closed the topic instead of simply posting a comment, so I've reopened it in case anyone else wants to contribute.

It works for me! thanks

suaveUT, 2022-07-08:
@fjs-acc -- I am facing the same issue. The link you have provided for pull request is not working. Can you tell me what do I need to change in the package.xml in order to install cartographer.

joaopeixoto13, 2022-07-08:
@suaveUT Remove the line: <depend>libabsl-dev</depend>

fjs-acc, 2022-07-08:
@suaveUT I didn't format the link properly, put you can just copy and past the adress that's actually there to reach the issue i was trying to link to.

You have to remove (delete/comment out, whatever floats your goat) line 46 (`<depend>libabsl-dev</depend>`) from the package.xml file in the cartographer (not cartographer ros!) package.