cartographer-project/cartographer_ros #1734: No/wrong trajectory in map

sai2602, 2022-07-26:
Hello,

I am new to cartographer and have been trying to generate a map for my use case. I have a VLP-16 lidar and using the VLP16_points.launch file in velodyne_pointcloud package, I generated a 2D laserscan. Although my lidar is moving (attached to a vehicle), the trajectory in the map doesn't move and submaps are just overwritten in the same place.

I tried tuning the parameters as mentioned in the documentation and noticed that changing min_range to negative values moves the map but the trajectory is wrong. 

I would like to know why changing min_range alone is impacting the map (but no other parameters are) and how do I tune it to my use case? 

My files are attached in the zip.

[cartographer.zip](https://github.com/cartographer-project/cartographer_ros/files/9188725/cartographer.zip)

eithwa, 2022-08-02:
[image]
Your laser data is insufficient, most of the time you can only get an open area with a radius of 6 meters

sai2602, 2022-08-02:
Hello,

Thanks for the update. When you say laser data is insufficient, do you mean the number of range points in the laser scan or the number of laser scan messages in the bag file? I think the number of range points in a message is not a problem as I tested this use case with a shorter distance scenario (I took a scan of my room while holding the vlp-16 and walking around) and this turned out to be fine.

eithwa, 2022-08-10:
> Thanks for the update. When you say laser data is insufficient, do you mean the number of range points in the laser scan or the number of laser scan messages in the bag file? I think the number of [...]

Because you only take part of the vlp-16 point cloud data into scan format, the maximum visible distance is only 6 meters. When you use it in outdoor scenes, the available range points are very few, because most objects are far beyond the visible range. If you can record the bag file with full point cloud2 data, I can show you what the correct 2d transformation should look like

sai2602, 2022-08-11:
> Because you only take part of the vlp-16 point cloud data into scan format, the maximum visible distance is only 6 meters. When you use it in outdoor scenes, the available range points are very few, because most objects are far beyond the visible range. You can refer to this for point cloud transformation https://github.com/ros-drivers/velodyne/tree/master/velodyne_laserscan

I used the same github repo for producing my laserscans from pointclouds. I have attached a sample bag file here for your reference. I have extracted ring 1 from the pointcloud.

[https://drive.google.com/drive/u/1/my-drive/bag_file.zip](url)

eithwa, 2022-08-12:
[image]

You can use point cloud data directly
https://gist.github.com/eithwa/406c4a7c65549b58a9682b70f8085a43

sai2602, 2022-08-16:
[image]

cartographer_node is not getting any input from my bag file and I getting a message that no publisher is currently active. I also re-recorded the bag file with the topic name remapped to points2 in the bag file (assuming the remap option in launch file did not work for some reason) and ran cartographer (with the remap now removed in the launch file) and it still throws the same error.

eithwa, 2022-08-23:
> Sorry for the delayed response. Working with pointclouds now generates better maps than the laserscan, but the initial problem of overwritten submaps still persists when the vehicle is moving fast. The improved results are only when the vehicle slows down or moves into a confined space like the parking lot. Is this something to do with motion filter settings or does 2D map generator expect truncated distance values?
> 
> [image]
> 
> The highlighted section is where the car was in a fast straight line and the maps were overwritten. After the right turn, the vehicle is relatively slower and the map is fine there.
> 
> Bag file for reference.
> 
> [https://drive.google.com/file/d/1YFcwxhkq7_FP6nfWU3eFCfD8CH1sCNZY/view?usp=sharing](url)

You never accepted my request for access to the file, so I can't help you.
The red circle in the picture is because the obstacle is beyond your laser limit, so there is not enough information to go to SLAM

sai2602, 2022-08-24:
I have made the file public. You should be able to access it now. The pointcloud does have data within range (some people, trees, cars and buildings).

eithwa, 2022-08-29:
Looks like it's hard to calculate visual odometry when moving fast, you could try adding gps, imu or odometry to assist slam