cartographer-project/cartographer_ros #1738: Cartographer doesn't be worked with Livox Mid-70 and IMU

yuataka, 2022-08-22:
Hi, I am trying to run cartographer with Livox Mid-70 and IMU.
I wrote lua file and launch file based on the following url issue, but it does not work.
https://github.com/cartographer-project/cartographer_ros/issues/1715

Cartographer_node doesn't publish submap and tf from map to base_link.

Could someone give me some advice? Thanks.

They are launch file, lua file and urdf file.
https://gist.github.com/yuataka/2bbad92d4878d5ae1d83847cb0a07ed4

This is bag file link.
https://drive.google.com/file/d/1GRIkVhWpX0Lch0aknlKNwid-QUU3Aujv/view?usp=sharing

This is rqt_graph result

[image]

This is rosbag_validate result.
```
E0822 17:42:41.280256 114269 rosbag_validate_main.cc:351] frame_id "livox_frame" on topic /livox/lidar has serialization time 1661157521.303560201 but sensor time 212.398777240 differing by -1.66116e+09 s.
E0822 17:42:41.280777 114269 rosbag_validate_main.cc:351] frame_id "livox_frame" on topic /livox/lidar has serialization time 1661157521.404656078 but sensor time 212.498617240 differing by -1.66116e+09 s.
E0822 17:42:41.281168 114269 rosbag_validate_main.cc:351] frame_id "livox_frame" on topic /livox/lidar has serialization time 1661157521.504676663 but sensor time 212.598457240 differing by -1.66116e+09 s.
I0822 17:42:41.426472 114269 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/imu" (frame_id: "imu_link"):
Count: 4136  Min: 5.115e-06  Max: 0.0104448  Mean: 0.00910518
[0.000005, 0.001049)	                    	Count: 88 (2.12766%)	Total: 88 (2.12766%)
[0.001049, 0.002093)	                    	Count: 76 (1.83752%)	Total: 164 (3.96518%)
[0.002093, 0.003137)	                    	Count: 81 (1.95841%)	Total: 245 (5.9236%)
[0.003137, 0.004181)	                    	Count: 72 (1.74081%)	Total: 317 (7.66441%)
[0.004181, 0.005225)	                    	Count: 75 (1.81335%)	Total: 392 (9.47776%)
[0.005225, 0.006269)	                    	Count: 75 (1.81335%)	Total: 467 (11.2911%)
[0.006269, 0.007313)	                    	Count: 75 (1.81335%)	Total: 542 (13.1044%)
[0.007313, 0.008357)	                    	Count: 82 (1.98259%)	Total: 624 (15.087%)
[0.008357, 0.009401)	                    	Count: 84 (2.03095%)	Total: 708 (17.118%)
[0.009401, 0.010445]	   #################	Count: 3428 (82.882%)	Total: 4136 (100%)
E0822 17:42:41.426533 114269 rosbag_validate_main.cc:383] Point data (frame_id: "livox_frame") has a large gap, largest is 0.1008 s, recommended is [0.0005, 0.05] s with no jitter.
I0822 17:42:41.426564 114269 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/livox/lidar" (frame_id: "livox_frame"):
Count: 375  Min: 0.09984  Max: 0.1008  Mean: 0.0999986
[0.099840, 0.099936)	   #################	Count: 313 (83.4667%)	Total: 313 (83.4667%)
[0.099936, 0.100032)	                    	Count: 0 (0%)	Total: 313 (83.4667%)
[0.100032, 0.100128)	                    	Count: 0 (0%)	Total: 313 (83.4667%)
[0.100128, 0.100224)	                    	Count: 0 (0%)	Total: 313 (83.4667%)
[0.100224, 0.100320)	                    	Count: 0 (0%)	Total: 313 (83.4667%)
[0.100320, 0.100416)	                    	Count: 0 (0%)	Total: 313 (83.4667%)
[0.100416, 0.100512)	                    	Count: 0 (0%)	Total: 313 (83.4667%)
[0.100512, 0.100608)	                    	Count: 0 (0%)	Total: 313 (83.4667%)
[0.100608, 0.100704)	                    	Count: 0 (0%)	Total: 313 (83.4667%)
[0.100704, 0.100800]	                 ###	Count: 62 (16.5333%)	Total: 375 (100%)

```

eithwa, 2022-08-23:
[image]
Your sensor's timestamps are out of sync
btw I think your point cloud data is not enough for slam

yuataka, 2022-08-23:
Thank you for your advice. I'll check timestamps.
About the point cloud,  is the lack of data temporal or spatial?

yuataka, 2022-08-24:
The cause is in livox timestamp and I need to set up ptp master as described in the Livox manual.
https://github.com/Livox-SDK/Livox-SDK/wiki/livox-device-time-synchronization-manual

I ran ptp master by this command.
'''
sudo ptpd -M -i eth0 -C
'''

After that, I could run cartographer with Livox and IMU.

About the lack of point cloud data is maybe the other problem.
So, I close this issue.

Thanks so much.

closed 2022-08-24 by yuataka

eithwa, 2022-08-24:
> Thank you for your advice. I'll check timestamps. About the point cloud, is the lack of data temporal or spatial?

I think that the spatial data of the sensor point cloud is insufficient. For slam, I think at least 180-degree information in the horizontal direction is needed, or it is helpful to provide additional odometer information.

yuataka, 2022-08-25:
Thanks for your reply. I'll try other 180-degree LiDAR.
Thanks so much.