cartographer-project/cartographer_ros #1741: while compile cartographer_ros for my own catkin_ws, it could not build "ros_log_sink.cc.o" , how to solve it? please

SCUT-LuMing, 2022-09-06:
==> ninja -j16 -l16 in '/home/luming/project/catkin_google_ws/build_isolated/cartographer_ros'
[3/41] Building CXX object CMakeFil.../cartographer_ros/ros_log_sink.cc.o
FAILED: CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_log_sink.cc.o 
/usr/bin/c++ -DGFLAGS_IS_A_DLL=0 -DGLOG_CUSTOM_PREFIX_SUPPORT -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -DURDFDOM_HEADERS_HAS_SHARED_PTR_DEFS -I/home/luming/project/catkin_google_ws/build_isolated/cartographer_ros -I/home/luming/project/catkin_google_ws/src/cartographer_ros/cartographer_ros -I/usr/src/googletest/googlemock/gtest/include -isystem /usr/include/lua5.2 -isystem /usr/local/include/pcl-1.9 -isystem /usr/local/include/eigen3 -isystem /home/luming/project/catkin_google_ws/install_isolated/include -isystem /opt/ros/melodic/include -isystem /opt/ros/melodic/share/orocos_kdl/cmake/../../../include -isystem /opt/ros/melodic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /usr/include/eigen3 -isystem /usr/include/pcl-1.8 -isystem /usr/include/ni -isystem /usr/include/openni2 -isystem /usr/include/vtk-6.3 -isystem /usr/include/freetype2 -isystem /usr/lib/x86_64-linux-gnu/openmpi/include/openmpi -isystem /usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent -isystem /usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent/include -isystem /usr/lib/x86_64-linux-gnu/openmpi/include -isystem /usr/include/python2.7 -isystem /usr/include/hdf5/openmpi -isystem /usr/include/libxml2 -isystem /usr/include/jsoncpp -isystem /usr/include/tcl -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/libpng16 -O3 -DNDEBUG  -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -march=native -msse4.2 -mfpmath=sse -MD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_log_sink.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_log_sink.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_log_sink.cc.o -c /home/luming/project/catkin_google_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_log_sink.cc
/home/luming/project/catkin_google_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_log_sink.cc: In member function ‘virtual void cartographer_ros::ScopedRosLogSink::send(google::LogSeverity, const char*, const char*, int, const tm*, const char*, size_t)’:
/home/luming/project/catkin_google_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_log_sink.cc:48:75: error: no matching function for call to ‘cartographer_ros::ScopedRosLogSink::ToString(const LogSeverity&, const char*, const int&, const tm* const&, const char* const&, const size_t&)’
       severity, GetBasename(filename), line, tm_time, message, message_len);
                                                                           ^
In file included from /home/luming/project/catkin_google_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_log_sink.h:22:0,
                 from /home/luming/project/catkin_google_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_log_sink.cc:17:
/usr/local/include/glog/logging.h:1788:22: note: candidate: static std::__cxx11::string google::LogSink::ToString(google::LogSeverity, const char*, int, const google::LogMessageTime&, const char*, size_t)
   static std::string ToString(LogSeverity severity, const char* file, int line,
                      ^~~~~~~~
/usr/local/include/glog/logging.h:1788:22: note:   no known conversion for argument 4 from ‘const tm* const’ to ‘const google::LogMessageTime&’
[14/41] Building CXX object CMakeFi...artographer_ros/msg_conversion.cc.o
/home/luming/project/catkin_google_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/msg_conversion.cc: In function ‘std::unique_ptr<nav_msgs::OccupancyGrid_<std::allocator<void> > > cartographer_ros::CreateOccupancyGridMsg(const cartographer::io::PaintSubmapSlicesResult&, double, const string&, const ros::Time&)’:
/home/luming/project/catkin_google_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/msg_conversion.cc:313:19: warning: variable ‘now’ set but not used [-Wunused-but-set-variable]
   const ros::Time now = ros::Time::now();
                   ^~~
[18/41] Building CXX object CMakeFi..._ros.dir/cartographer_ros/node.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/luming/project/catkin_google_ws/install_isolated/env.sh', 'ninja', '-j16', '-l16']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/luming/project/catkin_google_ws/build_isolated/cartographer_ros && /home/luming/project/catkin_google_ws/install_isolated/env.sh ninja -j16 -l16

MigYang, 2022-10-23:
I'm same error,and i succeed to install.
Error reason: glog version incompatible.(my glog version is 0.7, cartographer_ros need version is 0.5.0).
Solution:

1. `sudo apt-get remove libgoogle-glog-dev` #uninstall current glog
2. `git clone https://github.com/google/glog.git`
3. `sudo apt-get install autoconf automake libtool`
4. `cd glog`
5. `mkdir build && cd build`
6. `cmake ..`
7. `make`
8. `sudo make install`
9. compile cartographer_ros in your own workspace

Attention:

1. cartographer version: 1.0.0
2. cartographer_ros version: 1.0.0
3. ceres-solver version: 1.13.0

SCUT-LuMing, 2022-10-23:
已收到你的来信，辛苦了！ have received your letter.Thank you.

calcualatexzy, 2023-04-04:
sorry but, your steps by cloning on github will automatically install a version 0.7.0 glog?
how can i change the version?

meturing38, 2024-10-08:
> sorry but, your steps by cloning on github will automatically install a version 0.7.0 glog? how can i change the version?

After cloning, you can check to new branch first, v0.5.0 or else.

traversaro, 2024-12-19:
A fix is proposed in https://github.com/ros2/cartographer_ros/pull/76 .