cartographer-project/cartographer_ros #1744: Cartographer on Jetson Nano

Cristian-wp, 2022-09-16:
Hello people,
I was thinking to run cartographer on a NVIDIA Jetson Nano 4Gb, but I have some questions about memory management.
From a lot of experiments on an Intel NUC with an i7 8core (4 physical and 4 virtual) I have see that this slam need a lot of cpu power to work (all my core were saturated during slam).
Recently I have discover that Ceres Solver can use a NVIDIA GPU in order to speed up to speed up DENSE_QR, DENSE_NORMAL_CHOLESKY and DENSE_SCHUR (http://ceres-solver.org/installation.html).
Do you have any experience about that?

adiego73, 2022-11-16:
I have used cartographer library (not the ROS node) on a Nvidia TX2 (8 GB, 4 cores), and it worked without problems. I have also compiled Ceres with GPU support and modified cartographer to use the following option:
`options.dense_linear_algebra_library_type = ceres::DenseLinearAlgebraLibraryType::CUDA;`

However, the GPU does not seem to be used.. I have used jtop to see the GPU usage, and the result were peaks of usage from time to time, but those peaks happend also when cartographer was not used at all.. I haven't seen any difference in performance. 

Maybe I am missing something, but those were the results I have obtained.

Cristian-wp, 2022-11-17:
Hi @adiego73 I have stop use cartographer, but recently I get your same behavior on another project. Maybe I am missing something too. I have ask to Ceres github page, and they say to look at the solver summary.