cartographer-project/cartographer_ros #1769: catkin_make_isolated --install --use-ninja giving error

kajariahridik, 2023-03-02:
<== Failed to process package 'cartographer': 
  Command '['/home/hridik/catkin_ws/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1.

How do I solve this error?

closed 2023-03-06 by kajariahridik

Abhirup0209, 2023-03-19:
Did you actually solve the error?

kajariahridik, 2023-03-20:
I did not solve it exactly, but I reinstalled ubuntu and started from scratch

Abhirup0209, 2023-03-20:
And did it work?

I am getting this error:

FAILED: docs/CMakeFiles/build_doc 
cd /home/royabhi02/catkin_gc_ws/build_isolated/cartographer/install/docs && /usr/bin/sphinx-build -b html /home/royabhi02/catkin_gc_ws/src/cartographer/docs/source /home/royabhi02/catkin_gc_ws/build_isolated/cartographer/install/docs/html
Running Sphinx v1.8.5

Extension error:
Could not import extension sphinx.builders.latex (exception: cannot import name 'contextfunction' from 'jinja2' (/home/royabhi02/.local/lib/python3.8/site-packages/jinja2/__init__.py))
[327/333] Building CXX object CMakeFiles/cartographer.sensor.compressed_point_cloud_test.dir/cartographer/sensor/compressed_point_cloud_test.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['ninja', '-j8', '-l8']' returned non-zero exit status 1.

yoggi56, 2023-08-16:
I've got the same error :(

FPSychotic, 2023-09-22:
same error , why is closed?

jhpark98, 2024-07-03:
Based on this [comment](https://github.com/readthedocs/readthedocs.org/issues/9037), I updated jinja2 to 3.0.0.
` pip install jinja2==3.0.0 `
It compiled magically!