cartographer-project/cartographer_ros #1779: 如何使用cartographer进行基于已有地图进行建图

CYALJH, 2023-04-13:


georgflick, 2023-04-24:
Hi @CYALJH, please describe your issue in english.

CYALJH, 2023-04-26:
> 嗨，请用英语描述您的问题。

hi,I want to say how to use cartographer to build map on existing maps.

IaroslavS, 2023-08-16:
> hi,I want to say how to use cartographer to build map on existing maps.

You mean how to merge several LiDAR maps ?

CYALJH, 2023-08-16:
No ，bro. It's not a merging of maps, it's an extension of an already existing map
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2023年08月16日 21:21 |
| 收件人 | ***@***.***> |
| 抄送人 | CYALJH ***@***.***>,
Mention ***@***.***> |
| 主题 | Re: [cartographer-project/cartographer_ros] 如何使用cartographer进行基于已有地图进行建图 (Issue #1779) |

hi,I want to say how to use cartographer to build map on existing maps.

You mean how to merge several LiDAR maps ?

yujun2019, 2024-05-15:
<node name="cartographer_offline_node" pkg="cartographer_ros"
      type="cartographer_offline_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files    # 配置文件路径
          -configuration_basenames xxxxxx.lua               # 配置文件名称
          -bag_filenames $(arg bag_filenames)
          -load_state_filename $(arg load_state_filename)"            
      output="screen">

yujun2019, 2024-05-15:
<node name="cartographer_offline_node" pkg="cartographer_ros"
      type="cartographer_offline_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files    # 配置文件路径
          -configuration_basenames xxxxx.lua               # 配置文件名称
          -bag_filenames $(arg bag_filenames)
          -load_state_filename $(arg load_state_filename)"            
      output="screen">