cartographer-project/cartographer_ros #1793: Localization-only mode still mapping

ChoiJ0527, 2023-09-06:
I am currently working on a multi-robot project. One of the issues we are running into is that if two robots are near each other and one robot is still for a while, the other robot will map the still robot into the global map.

To fix this, I tried to change my .lua configuration and run Cartographer in localization-only mode as [documented here](https://google-cartographer-ros.readthedocs.io/en/latest/going_further.html#localization-only):

```
TRAJECTORY_BUILDER.pure_localization_trimmer = {
    max_submaps_to_keep = 3,
}
```

Even after this modification, Cartographer will still keep mapping and adding to the global map. A change that I did notice is the global map will reset to it's original state if the robot moves to (what I'm assuming is) a different submap from where the modifications occurred, but I am hoping that we don't have to do this.

Is there any way to prevent mapping completely, either by tuning parameters or by changing the code?

eithwa, 2023-09-27:
https://github.com/cartographer-project/cartographer_ros/blob/c138034db0c47fe0ea5a2abe516acae02190dbf5/cartographer_ros/cartographer_ros/occupancy_grid_node_main.cc#L45C39-L45C43

Maybe you could try making it output only the frozen submaps

lucamozza, 2023-09-29:
The way localization mode works in cartographer is to start a new trajectory.
This means that you will have at least two trajectories in the pose graph: one map trajectory (which is frozen and will not be optimized) and the live robot trajectory.
The latter only keeps the last N (3 in your case) submaps, dropping the old ones.
I think this is the behavior that you are experiencing: once new submaps are created the old one depicting your robot get eliminated.
There is no way to avoid mapping altogether, since this is the fundamental behavior of Cartographer.
Also, I would say that philosophically this behavior is not wrong: the robot standing sill is, temporarily, a useful landmark to consider.

A note on @eithwa 's answer: that is correct if you only care about visualization of the map. The acutual behavior of Cartographer will not be influenced by the setting of the occupancy grid node. It is, after all, a different node.

mertcanduz, 2023-10-03:
> The way localization mode works in cartographer is to start a new trajectory. This means that you will have at least two trajectories in the pose graph: one map trajectory (which is frozen and will [...]

> The way localization mode works in cartographer is to start a new trajectory. This means that you will have at least two trajectories in the pose graph: one map trajectory (which is frozen and will [...]

Hey,

I'm using Cartographer for pure localization and Cartographer is mapping when the pure-localization mode is open. If I use the map_server node to publish the map and I don't use cartographer_grid_node, would it be better? According to what you said, there will be no change. What do you think?

lucamozza, 2023-10-03:
> Hey,
> 
> I'm using Cartographer for pure localization and Cartographer is mapping when the pure-localization mode is open. If I use the map_server node to publish the map and I don't use cartographer_grid_node, would it be better? According to what you said, there will be no change. What do you think?

I do not understand the goal you want to achieve.
The map_server node is not capable of publishing cartographer maps, unless you create them with the occupancy grid node, save them and republish them.

Could you elaborate on your question?

mertcanduz, 2023-10-03:
Sure,

Firstly, I created a map with a cartographer. Then I saved the map as a PNG file using the map_server node. Now I'm using the map_server node instead of cartographer_occupancy_grid_node. I wonder if this has any effect on the Cartographer's background mapping. I can give an example like this:

the old launch file is :

<node name="cartographer_node" pkg="cartographer_ros"
    type="cartographer_node" args="
        -configuration_directory $(find my_navigation)/configuration_files
        -configuration_basename 2d_localization.lua
        -map_filename $(arg map_filename)"
    output="screen">
</node>

<node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
    type="cartographer_occupancy_grid_node" args="-resolution 0.05" /> 

the new launch file is:

<node name="cartographer_node" pkg="cartographer_ros"
    type="cartographer_node" args="
        -configuration_directory $(find my_navigation)/configuration_files
        -configuration_basename 2d_localization.lua
        -map_filename $(arg map_filename)"
    output="screen">
</node>

<node name="map_server" pkg="map_server" type="map_server" args="$(find my_navigation)/map.yaml">  
      <param name="frame_id" value="map"/>
</node>

lucamozza, 2023-10-03:
It does not have any effect on Cartographer mapping process.
Cartographer does not use png maps to localize, but rather saves the full pose graph+occupancy grids and more in a pbstream file.
The occupancy_grid_node  is just visualization / export tool.

I also notice that you use the `-map_filename` flag, which is not a supported flag for the cartographer node. Are you able to run those launch files without crashes?

mertcanduz, 2023-10-03:
Hi again,

Of course, I can not run this launch file. I just want to give an example. My launch file is not like this.

Thank you for your information. I got this point. There will be no difference and I want to ask a question. You said, " I would say that philosophically this behavior is not wrong: the robot standing sill is, temporarily, a useful landmark to consider. ". However, my robot has a path where it constantly turns, and since the map constantly updates itself after the robot works for 3-4 hours, the map and therefore the localization shifts by 30-40 degrees. How can I prevent this situation? Do you have any advice?

eithwa, 2023-10-03:
>However, my robot has a path where it constantly turns, and since the map constantly updates itself after the robot works for 3-4 hours, the map and therefore the localization shifts by 30-40 degrees. How can I prevent this situation? Do you have any advice?

I think the parameters you are using may not be allowing the new submap to match well with the old one, which could be causing this issue. You can try adjusting these parameters to make it easier for them to match each other, but of course, this is just one possible guess.
```
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.5
POSE_GRAPH.constraint_builder.min_score = 0.4
```

lucamozza, 2023-10-03:
Since this is another possibly lengthy topic i suggest you open another issue :)

mertcanduz, 2023-10-03:
Thank you for your advice @eithwa. I'll try to set these parameters.