cartographer-project/cartographer_ros #1798: After long time, the map is changing.

mcdrob, 2023-10-03:
Hi!

I'm using Cartographer for localization and mapping. However, my robot has a path where it constantly turns, and since the map constantly updates itself after the robot works for 3-4 hours, the map and therefore the localization shifts by 30-40 degrees. How can I prevent this situation? Do you have any advice?

Some params I use are like these : 

    TRAJECTORY_BUILDER.pure_localization_trimmer = {
      max_submaps_to_keep = 4,
    }
    POSE_GRAPH.constraint_builder.global_localization_min_score =  0.5
    POSE_GRAPH.constraint_builder.min_score = 0.4

eithwa, 2023-10-04:
If you can provide bag, lua, pbstream, launch and urdf files I can help you test it. I am interested in the long-term movement situation.

naitiknakrani-eic, 2024-09-10:
@mcdrob Did you get your solution ? I am facing the same issue.

mcdrob, 2024-09-10:
No, I didn't. I'm using other packages(algorithms) for localization. I think a cartographer is not suitable for that.

naitiknakrani-eic, 2024-09-10:
Which one ? Is it ROS based ?