cartographer-project/cartographer_ros #1804: How to configure Cartographer_ROS to create new submaps when trajectory path elevates on Z axis?

amalhub, 2023-11-24:
I have configured the cartographer_ros to run with my Ouster LiDAR according to configurations given in: 
[ouster_example/cartographer_ros](https://github.com/wilselby/ouster_example/blob/master/cartographer_ros/configuration_files/cart_3d.lual).

When I run the realtime SLAM using the my LiDAR it starts to generate the ground level map perfectly and when I climb the stairs the trajectory path elevates along with the pointcloud data but it doesn't create a new submap at upper level, instead it starts to override (/overlay) the ground level map with new occupancy grid data. Below RVIZ screenshot shows the issue: 

[image]

Does anyone know what parameters effect or what parameters I need to change to get it creating new submaps when the trajectory path elevates to a new level (/floor)? 

The default demo that provided by the [Cartographer official documentation](https://google-cartographer-ros.readthedocs.io/en/latest/demos.html) works well as shown in below screenshot. I need similar results with my Ouster LiDAR configurations. I tried copying the default config parameters to my config files but it had no luck. Also I tried checking the documentations to figure out the settings but had no luck and it is overwhelming due to so many configuration parameters. Any help to figure out what configurations affect the new submap creation when the trajectory elevates is much appreciated. 

[image]

adiego73, 2023-12-05:
Hi, I haven't use the ROS node, but I have used Cartographer stand-alone. I don't think you can map multiple floors on the same map. I remember reading something about **_flat floor assumption_**, which means cartogrpaher assumes there won't be any ramp or steps, or in your case, another floor on the same map.

A way to solve this issue may be to have multiple maps (one per floor) and reinitialize cartographer any time you know you have moved from one floor to another. 

Hope this helps.

amalhub, 2023-12-06:
Hi @adiego73, 
Thank you for your prompt response. I appreciate your input; however, I find myself unable to fully align with your explanation. In my experimentation with the demo data outlined in the [Cartographer documentation](https://google-cartographer-ros.readthedocs.io/en/latest/demos.html), I observed the LiDAR capturing a person climbing stairs at the "Deutsches Museum". In the context of Cartographer's ***3D SLAM***, the results showcased the creation of multiple sub-maps within a singular RVIZ map, as evidenced by the screenshot attached to my initial message.

I encourage you to run the provided demo data and witness these outcomes firsthand by following the guidelines outlined in the documentation. Based on my observations, I think it can create multiple floor maps on the same map, and I think it has to have some configuration parameter that starts creating a new submap when a particular threshold parameter exceeds. I think in my settings, it seems the threshold parameter (default) is set too high, leading to the omission of generating new submaps for smaller elevations, resulting in overriding the initial map.

I would greatly appreciate any help finding this configuration parameter.