cartographer-project/cartographer_ros #1810: Is the compression/expansion of a line scan when rotating taken into account?

alejandronespereira, 2024-01-30:
I am trying to improve the results I am getting from ros_cartographer using a scan lidar. When inspecting the raw scan data I have observed that parallel walls from a corridor appear parallel on the scan if I am moving in straight lines, but that is not the case when I rotate the scan; walls are not straight nor parallel between them. As far as my understanding goes this is to be expected, as the laser that gets the measurements is rotating on its own. 

The scan message has a time and angle increment between each measurement. 

My question is, does cartographer fix this by using the data or do I need to pre-process it myself. In case it does, does it need something? I am currently supplying cartographer with a visual odometry topic and an IMU. The maps that I am generating have some flaws, but on Rviz the walls from the scan appear straight, but I would like to have official confirmation, and I could not find anything on the docs. 

Thanks

zjlian, 2024-02-03:
You can take a look at the `num_subdivisions_per_laser_scan` and `num_accumulated_range_data` in the configuration file, as they are used for correcting motion distortion.

alejandronespereira, 2024-02-06:
So I understand that cartographer does correct the motion distortion. I will look into those, thanks!

closed 2024-02-06 by alejandronespereira