cartographer-project/cartographer_ros #1818: Cartographer Map obstacles keep disappearing

wbzi, 2024-04-08:
Hi, I am trying to map an area in which there are desks and chairs. When I walk in one direction the tables are mapped as obstacles, however when i turn around to go around the area, the tables disappear. I have tried changing the parameters but to no avail. What steps can I take to solve this issue?

marmik18, 2024-11-12:
I have a similar issue, can someone help with this.
Thank you

marmik18, 2024-11-12:
Hey @wbzi 
It is mentioned [here](https://github.com/cartographer-project/cartographer_ros/issues/1457) that the disappearing not explicitly modeled but rather part of the probability grid generatation. 

I was able to tweak it by playing around with these 2 values in my `.lua` configuration:
```
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.hit_probability=0.75
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.miss_probability=0.49
```

The complete configuration can be found [here](https://github.com/cartographer-project/cartographer/blob/master/configuration_files/trajectory_builder_2d.lua).