cartographer-project/cartographer_ros #1819: ros melodic Compilation failure

Fxx0105, 2024-04-08:
The following is the error messages：

FAILED: CMakeFiles/cartographer.mapping.internal.constraints.constraint_builder_3d_test.dir/cartographer/mapping/internal/constraints/constraint_builder_3d_test.cc.o 
/usr/bin/c++  -DGFLAGS_IS_A_DLL=0 -I/home/fu/cartographer_ws/src/cartographer/cartographer -I. -I/home/fu/cartographer_ws/src/cartographer -isystem /usr/src/googletest/googlemock/include -isystem /usr/include/eigen3 -isystem /usr/include/lua5.2 -O3 -DNDEBUG    -pthread -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -pthread -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MD -MT CMakeFiles/cartographer.mapping.internal.constraints.constraint_builder_3d_test.dir/cartographer/mapping/internal/constraints/constraint_builder_3d_test.cc.o -MF CMakeFiles/cartographer.mapping.internal.constraints.constraint_builder_3d_test.dir/cartographer/mapping/internal/constraints/constraint_builder_3d_test.cc.o.d -o CMakeFiles/cartographer.mapping.internal.constraints.constraint_builder_3d_test.dir/cartographer/mapping/internal/constraints/constraint_builder_3d_test.cc.o -c /home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/constraints/constraint_builder_3d_test.cc
In file included from /home/fu/cartographer_ws/src/cartographer/cartographer/common/task.h:24:0,
                 from /home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/constraints/constraint_builder_3d.h:34,
                 from /home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/constraints/constraint_builder_3d_test.cc:17:
/home/fu/cartographer_ws/src/cartographer/cartographer/common/thread_pool.h:67:58: error: expected ‘;’ at end of member declaration
   std::weak_ptr<Task> Schedule(std::unique_ptr<Task> task)
                                                          ^
/home/fu/cartographer_ws/src/cartographer/cartographer/common/thread_pool.h:68:22: error: ‘mutex_’ has not been declared
       LOCKS_EXCLUDED(mutex_) override;
                      ^~~~~~
/home/fu/cartographer_ws/src/cartographer/cartographer/common/thread_pool.h:68:30: error: ISO C++ forbids declaration of ‘LOCKS_EXCLUDED’ with no type [-fpermissive]
       LOCKS_EXCLUDED(mutex_) override;
                              ^~~~~~~~
/home/fu/cartographer_ws/src/cartographer/cartographer/common/thread_pool.h:73:46: error: expected ‘;’ at end of member declaration
   void NotifyDependenciesCompleted(Task* task) LOCKS_EXCLUDED(mutex_) override;
                                              ^
/home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/2d/pose_graph_2d.h:290:71: error: expected ‘;’ at end of member declaration
     void SetTrajectoryState(int trajectory_id, TrajectoryState state) override
                                                                       ^~~~~~~~
/home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/2d/pose_graph_2d.h:291:34: error: ‘parent_’ has not been declared
         EXCLUSIVE_LOCKS_REQUIRED(parent_->mutex_);
                                  ^~~~~~~
/home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/2d/pose_graph_2d.h:291:41: error: expected ‘,’ or ‘...’ before ‘->’ token
         EXCLUSIVE_LOCKS_REQUIRED(parent_->mutex_);
                                         ^~
/home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/2d/pose_graph_2d.h:291:49: error: ISO C++ forbids declaration of ‘EXCLUSIVE_LOCKS_REQUIRED’ with no type [-fpermissive]
         EXCLUSIVE_LOCKS_REQUIRED(parent_->mutex_);
                                                 ^
/home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/2d/pose_graph_2d.h:291:9: error: ‘int cartographer::mapping::PoseGraph2D::TrimmingHandle::EXCLUSIVE_LOCKS_REQUIRED(int)’ cannot be overloaded
         EXCLUSIVE_LOCKS_REQUIRED(parent_->mutex_);
         ^~~~~~~~~~~~~~~~~~~~~~~~
/home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/2d/pose_graph_2d.h:281:9: error: with ‘int cartographer::mapping::PoseGraph2D::TrimmingHandle::EXCLUSIVE_LOCKS_REQUIRED(int)’
         EXCLUSIVE_LOCKS_REQUIRED(parent_->mutex_);
         ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/fu/cartographer_ws/src/cartographer/cartographer/mapping/internal/2d/pose_graph_2d_test.cc:24:0:
/home/fu/cartographer_ws/src/cartographer/cartographer/common/internal/testing/lua_parameter_dictionary_test_helpers.h: In member function ‘virtual std::__cxx11::string cartographer::common::DummyFileResolver::GetFileContentOrDie(const string&)’:
/home/fu/cartographer_ws/src/cartographer/cartographer/common/internal/testing/lua_parameter_dictionary_test_helpers.h:42:3: error: control reaches end of non-void function [-Werror=return-type]
   }
   ^
/home/fu/cartographer_ws/src/cartographer/cartographer/common/internal/testing/lua_parameter_dictionary_test_helpers.h: In member function ‘virtual std::__cxx11::string cartographer::common::DummyFileResolver::GetFullPathOrDie(const string&)’:
/home/fu/cartographer_ws/src/cartographer/cartographer/common/internal/testing/lua_parameter_dictionary_test_helpers.h:46:3: error: control reaches end of non-void function [-Werror=return-type]
   }
   ^
cc1plus: some warnings being treated as errors
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['ninja', '-j20', '-l20']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/fu/cartographer_ws/build_isolated/cartographer && ninja -j20 -l20

Command failed, exiting.

I was hoping someone could help me out

Fxx0105, 2024-04-08:
哪位大佬知道或是遇到过这种错误？劳烦指点一下，感谢

liarokapisv, 2024-08-02:
This is due to an older `abseil` version being required. The macros are defined in `abseil`'s `common/thread_annotations.h` header, but after 2019, `abseil` adds an `ABS_` prefix to them.

Fxx0105, 2024-08-03:
Thank you very much for your answer
Thank you very much for your answer

fxx.
***@***.***

Fxx0105, 2024-08-03:
> This is due to an older `abseil` version being required. The macros are defined in `abseil`'s `common/thread_annotations.h` header, but after 2019, `abseil` adds an `ABS_` prefix to them.

Thank you so much for your answer，and i will change my `abseil` version later

shaoxh, 2025-01-08:
I solve it by checking out tag of absl to 20190808 and ceres to 2.0.0!
Great!