cartographer-project/cartographer_ros #1826: Demo bag files are not available to download

amalhub, 2024-05-30:
Hi, 

I have been following the ROS documentation (https://google-cartographer-ros.readthedocs.io/en/latest/demos.html) and tried to download the demo bag files but they are unavailable to download. 
Getting the below error when I run the wget command: 
```
--2024-05-30 16:45:40--  https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/b2-2016-04-05-14-44-52.bag
Resolving storage.googleapis.com (storage.googleapis.com)... 142.250.66.219, 172.217.167.123, 142.250.71.91, ...
Connecting to storage.googleapis.com (storage.googleapis.com)|142.250.66.219|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2024-05-30 16:45:41 ERROR 403: Forbidden.
```
When I try to directly download the file from the browser, I get the below error: 
```
<Error>
<Code>UserProjectAccountProblem</Code>
<Message>
The project to be billed is associated with a closed billing account.
</Message>
<Details>
The billing account for the owning project is disabled in state closed
</Details>
</Error>
```
Seems like either the demo bag files have been relocated and the urls are not updated in the documentation or the public access to the demo bag files have been revoked. 
Can you please send me the correct urls to download the bag files or update the documentation links and let me know?

Leo6069, 2024-07-03:
Hi mate, I am facing the sam problem. Have you solved it? Cheers

amalhub, 2024-07-04:
> Hi mate, I am facing the sam problem. Have you solved it? Cheers

Nope. No response from anyone.

engels9527, 2024-07-13:
same problem.Can a handsome man send me a rosbag using landmark please ?
you are my hero ,thank you.i beg beg beg beg beg beg beg beg beg beg you