cartographer-project/cartographer_ros #1838: Error compiling on nvidia jetson xavier nx

zhang-mu-zhi, 2024-09-06:
I get the following error when compiling cartographer_ros on nvidia jetson xavier nx：

```
FAILED: docs/CMakeFiles/build_doc 
cd /home/nvidia/cartographer_ws/build_isolated/cartographer/install/docs &amp;&amp; /home/nvidia/anaconda3/bin/sphinx-build -b html /home/nvidia/cartographer_ws/src/cartographer/docs/source /home/nvidia/cartographer_ws/build_isolated/cartographer/install/docs/html
Running Sphinx v4.0.1

Exception occurred:
  File "/home/nvidia/.local/lib/python3.8/site-packages/pillow-10.4.0-py3.8-linux-aarch64.egg/PIL/_typing.py", line 10, in &lt;module&gt;
    NumpyArray = npt.NDArray[Any]
AttributeError: module 'numpy.typing' has no attribute 'NDArray'
The full traceback has been saved in /tmp/sphinx-err-3g0e9ube.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at &lt;https://github.com/sphinx-doc/sphinx/issues&gt;. Thanks!
[46/381] Building CXX object CMakeFiles/cartographer.transform.transform_test.dir/cartographer/transform/transform_test.cc.o
ninja: build stopped: subcommand failed.
&lt;== Failed to process package 'cartographer': 
  Command '['ninja', '-j4', '-l4']' returned non-zero exit status 1.

Reproduce this error by running:
==&gt; cd /home/nvidia/cartographer_ws/build_isolated/cartographer &amp;&amp; ninja -j4 -l4

Command failed, exiting.
nvidia@nvidia-desktop:~/cartographer_ws$ 
```
My system is Ubuntu 20.04, Jetpack 5.1.1.

zhang-mu-zhi, 2024-09-06:
What should I do to fix it?

marmik18, 2024-11-13:
please use https://github.com/ros2/cartographer_ros