cartographer-project/cartographer_ros #1844: running deutsches_museum error

Clushio, 2024-12-07:
any solution?
```
nvidia@ubuntu:~/catkin_ws$ roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=/home/nvidia/Downloads/cartographer_paper_deutsches_museum.bag
... logging to /home/nvidia/.ros/log/0474833e-b47a-11ef-9428-cc47406fb951/roslaunch-ubuntu-101753.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:35911/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: noetic
 * /rosversion: 1.17.0
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [101842]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0474833e-b47a-11ef-9428-cc47406fb951
process[rosout-1]: started with pid [101913]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [101917]
process[cartographer_node-3]: started with pid [101918]
process[cartographer_occupancy_grid_node-4]: started with pid [101922]
process[rviz-5]: started with pid [101923]
process[playbag-6]: started with pid [101927]
[ INFO] [1733562163.823808804]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1733562163.834722167]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1733562163.834822960]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1733562163.834921129]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1733562163.834968613]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1733562163.835144344]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1733562163.835186997]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1733562163.835265903]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1733562163.835305196]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1733562163.835513980]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1733562163.835574424]: I1207 17:02:43.000000 101918 configuration_file_resolver.cc:41] Found '/home/nvidia/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1733562163.890692146]: I1207 17:02:43.000000 101918 map_builder_bridge.cc:135] Added trajectory with ID '0'.
[ INFO] [1733562164.794091192, 1432647016.482139400]: I1207 17:02:44.000000 101918 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '635682438164904058'.
[ INFO] [1733562164.806312682, 1432647016.482139400]: I1207 17:02:44.000000 101918 pose_graph_2d.cc:148] Inserted submap (0, 0).
free(): invalid size
[cartographer_node-3] process has died [pid 101918, exit code -6, cmd /home/nvidia/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/nvidia/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/nvidia/.ros/log/0474833e-b47a-11ef-9428-cc47406fb951/cartographer_node-3.log].
log file: /home/nvidia/.ros/log/0474833e-b47a-11ef-9428-cc47406fb951/cartographer_node-3*.log
```

ChuangPeng, 2024-12-22:
meet same problem, have you solved it ?

Clushio, 2024-12-22:
> meet same problem, have you solved it ?

Sorry, I haven't solved this problem, but I tried to use docker to mount and run the compiled cartographer in the Melodic environment, and it magically succeeded.

ChuangPeng, 2024-12-22:
> > meet same problem, have you solved it ?
> 
> Sorry, I haven't solved this problem, but I tried to use docker to mount and run the compiled cartographer in the Melodic environment, and it magically succeeded.

oh, thank you for the comment, maybe ros1/ros2 cause this...

closed 2024-12-22 by Clushio