cartographer-project/cartographer_ros #1846: Map frame built based on a target frame

ngwk1, 2025-02-21:
Hi, 

I would like to create the map in the same 2D Plane as per a target frame. Currently, it seems to be building the map based on the `tracking_frame`. This is an issue because if the `tracking_frame` and the `lidar_frame` isn't aligned on the same plane, the map can't be used for navigation directly due to scan data being occluded and alignment issues.

TiNredmc, 2025-05-11:
I've tested that if I set tracking frame to something like base link. The cartographer still able to localize. Though you have to make sure that the transform between your lidar and base link are correct (be most precise as much as you can).

gitwk27, 2025-05-16:
I think the issue was because map is built using the tracking frame and the tracking frame (imu) is tilted. The whole map built is tilted instead of aligning flat with the x-y plane. This makes navigation and other related work challenging..

TiNredmc, 2025-05-16:
> I think the issue was because map is built using the tracking frame and the tracking frame (imu) is tilted. The whole map built is tilted instead of aligning flat with the x-y plane. This makes navigation and other related work challenging.. 

Yeah, you right. Since the 2D SLAM doesn't require IMU. Using base_link as tracking frame was fine. But in the case of 3D SLAM (that I just started using them). Using IMU frame as tracking is so troublesome that I probably have to do some magic tricks after looking up transform map to base_link