cartographer-project/cartographer_ros #1849: improve localization stability and avoid sudden pose jumps during navigation.

lmmir, 2025-07-19:
How can I modify the source code to prevent large localization jumps, or ensure that localization corrections are only applied within a 3-meter radius around the robot’s current pose? If the correction exceeds 3 meters, it should be discarded or ignored.

lmmir, 2025-08-05:
> motion_filter
i use 2d , and max_distance_meters is not work

ranabotics, 2026-09-03:
@lmmir  have you got the solution for your problem ? currently i am running into the same problem as you were, if you have done with that would you please share the idea ?

lmmir, 2026-09-08:
> [@lmmir](https://github.com/lmmir) have you got the solution for your problem ? currently i am running into the same problem as you were, if you have done with that would you please share the idea ?

@ranabotics Chinese or foreigner?